MotorDynamicsLab/Nitehawk-SB #4: Zero RPM when part cooling fan above 90%.

z00mantwo, 2024-02-19:
Is there an RPM limit for readings from the Nitehawk board? My part cooling fan reads RPM when below 90% above that it reads 0 rpm. The fan is capable of 5,500 RPM.  

Fan is
https://www.digikey.com/en/products/detail/delta-electronics/BFB0524H-R00/5799880

cneshi, 2024-02-20:
that is a very strange issue, I wonder if it has to do with the `tachometer_poll_interval` Can you tweak this value to see if anything changes Here is the reference for it: https://www.klipper3d.org/Config_Reference.html#fan

z00mantwo, 2024-02-20:
That didn't make a difference. I think maybe the fan does not really give RPM. Instead it gives a "running" signal. See

https://www.delta-fan.com/Download/Spec/BFB0524H-R00.pdf

Page 8. At least I think that is what it is saying. I may be wrong.

z00mantwo, 2024-02-20:
I tried this fan and RPM seems to be reading properly

https://www.digikey.com/en/products/detail/cui-devices/CBM-5015BF-055-391-20/19524615

cneshi, 2024-02-22:
> That didn't make a difference. I think maybe the fan does not really give RPM. Instead it gives a "running" signal. See
> 
> https://www.delta-fan.com/Download/Spec/BFB0524H-R00.pdf
> 
> Page 8. At least I think that is what it is saying. I may be wrong.

you are right, according to the datasheet this delta fan only gives a simple high/low signal indicating whether it is running or not.

closed 2024-02-22 by cneshi