MotorDynamicsLab/Nitehawk-SB #10: Lost communication with MCU 'nhk' Nitehawk keeps disconnecting

KastenTaco, 2024-08-22:
My Nitehawk keeps disconnecting about an hour into each print. 
At this point I have checked all connectors multiple times, made sure things don't overheat (47C max temperature), re-flashed Klipper, replaced the RPI, and tried a new USB cable. 
All to no avail. It seems somewhat random, and I am at a loss.

Are there any known hardware or quality issues with the board itself? This is the only component I can not test and verify.

R-yang1, 2024-08-23:
Hi, this is Rebecca from LDO team.
Could you please send a email to support@ldomotors.com with a picture of wiring and the klippy.log?

Thanks.

patlap13, 2024-12-24:
I had the same issue. a soon as toouch yhe cable I got the error

Daisy1819, 2024-12-26:
> I had the same issue. a soon as toouch yhe cable I got the error

Hi, Could u write an email to describe your problem to our support account, it should be support@ldomotors.com.

patlap13, 2024-12-26:
I found the problem actually

The connector is getting looser and Connor  stay un place

I’m designing a retainer fie the Galileo extruder.

Daisy1819, 2024-12-27:
Glad to hear that. If the connector is loose, i recommend u can use a tweezers to enlarge the connect ports, it may help.

Best regards,
Daisy

***@***.***
 
From: NurseKate
Date: 2024-12-26 18:01
To: MotorDynamicsLab/Nitehawk-SB
CC: Daisy1819; Comment
Subject: Re: [MotorDynamicsLab/Nitehawk-SB] Lost communication with MCU 'nhk' Nitehawk keeps disconnecting (Issue #10)
I found the problem actually 

The connector is getting looser and Connor stay un place 

I’m designing a retainer fie the Galileo extruder.

Daisy1819, 2024-12-27:
like the photo.

***@***.***
 
From: daisy
Date: 2024-12-27 14:36
To: MotorDynamicsLab/Nitehawk-SB; MotorDynamicsLab/Nitehawk-SB
CC: Comment
Subject: Re: Re: [MotorDynamicsLab/Nitehawk-SB] Lost communication with MCU 'nhk' Nitehawk keeps disconnecting (Issue #10)
Glad to hear that. If the connector is loose, i recommend u can use a tweezers to enlarge the connect ports, it may help.

Best regards,
Daisy

***@***.***
 
From: NurseKate
Date: 2024-12-26 18:01
To: MotorDynamicsLab/Nitehawk-SB
CC: Daisy1819; Comment
Subject: Re: [MotorDynamicsLab/Nitehawk-SB] Lost communication with MCU 'nhk' Nitehawk keeps disconnecting (Issue #10)
I found the problem actually 

The connector is getting looser and Connor stay un place 

I’m designing a retainer fie the Galileo extruder.

JJayzX, 2025-08-15:
I've been getting this issue intermittently with new board but is now suddenly constant. It is not the connection as it has happened while in heating phase, so no movement, and no matter what I move around does not cause it to disconnect. Mainboard is Manta M8P v2 with RPi 5 and all the software is updated to today.

z00mantwo, 2025-08-15:
I had a similar problem. Switching the Nitehawk to the USB 2.0 port on my Raspberry Pi 4b with the other USB 2.0 port empty has fixed it for me.

z00mantwo, 2025-09-04:
Try adding 

usbcore.autosuspend=-1

to the end of the line in your /boot/cmdline.txt or /boot/firmware/cmdline.txt depending on the version of the OS you have.

Has to be at the end of the single line that is in the cmdline.txt. With a space before it. This will prevent the OS from trying to save power by turning off USB devices. It has helped clear up issues with long prints for me. Hopefully this will help.

akm25-v24, 2026-03-16:
I have kind of same issue, lost communication and now middle of print, system shutdown. I use M8P V2.0 and CB2.

bigrich12478, 2026-03-24:
Hello,
I am also building a voron 2.4 with stealthchanger system using 5 Jabberwocky Toolheads and Nighthawk 36s (Ver. 1.5.1 USB board) toolhead board. The R6 and R7 resistors are not there. Temp is 40 degrees Celsius. I am losing connection early in the print. I am utilizing cable strain relief. Not sure where to look from here.
Full Setup:
BTT Manta M8P V2.0
BTT CB2
LDO Jabeberwocky complete toolhead kit with NH36 kits.

vsevolodgarkusha, 2026-10-03:
Hello. I have the same issue.
First time an issue was occurred after an hour of printing time, the second one is after three hours.

LDO 2.4 rev D
Nitehawk 1.5, leviathan 1.3, Raspberry PI 4 1GB

[Sat Oct  3 17:41:11 2026] usb 1-1.4: USB disconnect, device number 4
[Sat Oct  3 17:41:12 2026] usb 1-1.4: new full-speed USB device number 5 using xhci_hcd
[Sat Oct  3 17:41:12 2026] usb 1-1.4: New USB device found, idVendor=1d50, idProduct=614e, bcdDevice= 1.00
[Sat Oct  3 17:41:12 2026] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Sat Oct  3 17:41:12 2026] usb 1-1.4: Product: rp2040
[Sat Oct  3 17:41:12 2026] usb 1-1.4: Manufacturer: Klipper
[Sat Oct  3 17:41:12 2026] usb 1-1.4: SerialNumber: 4E363334320BA49B
[Sat Oct  3 17:41:12 2026] cdc_acm 1-1.4:1.0: ttyACM2: USB ACM device

[image]

[image]