MotorDynamicsLab/Nitehawk-SB #19: Nitehawk SB not recognized on Rpi 5 usb

ecksa, 2025-08-10:
I recently upgraded my rpi3b+ to a rpi 5. Nitehawk works without issue on rpi3 but it will not recognize it on rpi5. I've tried in boot mode and everything, it's like its not plugged in (i have checked the cable and made sure it was fully seated). Other usb devices are found no issue, just the nitehawk board. If i move the cable back to the rpi3 it finds it not problem, right away, so it's clearly not the NH board being burned out or something. I have tried to disable usb current limit. I am also using an official 5v 5a power supply for the pi.

z00mantwo, 2025-08-13:
From the Nitehawk, the Pi 5, or something else? I'm doing this upgrade myself soon.

ecksa, 2025-08-13:
[R7 Resistor Removal Guide.pdf](https://github.com/user-attachments/files/21764000/R7.Resistor.Removal.Guide.pdf)